hildogjr/KiCost #544: Kicost add custom parts and inventory from external file

matiasbarone, 2023-03-03:
I need to add components from some local suppliers and also add the components that I have in inventory. I need to add them in multiple projects.

What I did was put the information in the components, but I have to edit component by component to keep it updated.

A test that I did was to build another xml with this information and use the manf# to link them, "it works" but it adds that other XML as if it were another board and mixes the quantitys, if I put qty 0 in the "local_suppliers" board it breaks the spreadsheet

I would like to have an option to add the list of parts but in another external file and take them into account for the analysis but not count them as another board.

set-soft, 2023-03-19:
I don't fully understand the case. What are these components? Why aren't they part of the project? Why computing the total quantities is a problem? Why isn't this a separate project?
It looks quite confusing and too particular. Perhaps you can explain it better.

matiasbarone, 2023-03-20:
> I don't fully understand the case. What are these components? Why aren't they part of the project? Why computing the total quantities is a problem? Why isn't this a separate project? It looks quite confusing and too particular. Perhaps you can explain it better.

Hello. I will try to explain better what I want to do.

From the bill of materials, there are many components that I have at various local suppliers or have enough in my warehouse.
To add the local prices or the quantity that I have in the warehouse, I have to edit each component of each project and add the Custom Part Data with the local prices or the quantities stored. The problem is that every time the prices change, or the quantities change, I have to edit each one of the components on every proyect.

If there was the possibility of entering a database (xml, csv, whatever) with the information, I would only have to maintain a single file and add the same database every time I run kicost in any project, and I only have to keep a single file updated.

set-soft, 2023-03-21:
Hi @matiasbarone !

Did you take a look to the KiCad 7 database support? https://forum.kicad.info/t/kicad-the-case-for-database-driven-design/34621

It sounds like it will help you

closed 2023-03-21 by matiasbarone