hildogjr/KiCost #546: Error when calling kicost without args to start GUI

ChrisBietz, 2023-05-10:
#### Issue / Problem report
1. Add the KiCost version.
KiCost v1.1.16
2. Add a terminal error report
```
Traceback (most recent call last):
  File "C:\Program Files\KiCad\6.0\bin\Lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\KiCad\6.0\bin\Lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Program Files\KiCad\6.0\bin\Scripts\kicost.exe\__main__.py", line 7, in <module>
  File "C:\Program Files\KiCad\6.0\bin\Lib\site-packages\kicost\__main__.py", line 479, in main
    main_real()
  File "C:\Program Files\KiCad\6.0\bin\Lib\site-packages\kicost\__main__.py", line 410, in main_real
    kicost_gui(args.force_en_us)  # Use the user gui if no input is given.
TypeError: kicost_gui() missing 4 required positional arguments: 'files', 'configure_kicost_apis', 'command_line_api_options', and 'args'
```
3. Write the command used to call KiCost (or the graphical interface configuration);
`kicost`
4. One BoM to reproduce the error (with the EDA version).
No BoM necessary to reproduce

ns2gupta, 2023-06-16:
I am having the same issue, any update on this?

set-soft, 2023-06-21:
Hi @ns2gupta !
The problem is fixed in the git code, you can use it.
Or you can just provide an input file from the command line and things will work. The problem is when you don't provide an input file.

ns2gupta, 2023-06-21:
Hi, 
so i added the config file and the api key for mouser in the config.yaml file. however, it still gives WARNING:(WC017) Can't enable Mouser without a `key` (kicost.distributors - log.py:107)

`C:\Users\naman\Documents\KiCad\7.0>kicost -c "C:\Program Files\KiCad\7.0\.config\kicost\config.yaml"
WARNING:(WC017) Can't enable Digi-Key without a `client_id`, `client_secret` and `cache_path` (kicost.distributors - log.py:107)
WARNING:(WC017) Can't enable Elemen14 without a `key` (kicost.distributors - log.py:107)
WARNING:(WC017) Can't enable TME without a `token` and an `app_secret` (kicost.distributors - log.py:107)
WARNING:(WC017) Can't enable Nexar without a `client_id`, `client_secret` and `cache_path` (kicost.distributors - log.py:107)
Traceback (most recent call last):
  File "C:\Program Files\KiCad\7.0\bin\Lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\KiCad\7.0\bin\Lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\naman\Documents\KiCad\7.0\3rdparty\Python39\Scripts\kicost.exe\__main__.py", line 7, in <module>
  File "C:\Users\naman\Documents\KiCad\7.0\3rdparty\Python39\site-packages\kicost\__main__.py", line 479, in main
    main_real()
  File "C:\Users\naman\Documents\KiCad\7.0\3rdparty\Python39\site-packages\kicost\__main__.py", line 410, in main_real
    kicost_gui(args.force_en_us)  # Use the user gui if no input is given.
TypeError: kicost_gui() missing 4 required positional arguments: 'files', 'configure_kicost_apis', 'command_line_api_options', and 'args'

C:\Users\naman\Documents\KiCad\7.0>kicost --show_dist_list
WARNING:(WC017) Can't enable Digi-Key without a `client_id`, `client_secret` and `cache_path` (kicost.distributors - log.py:107)
WARNING:(WC017) Can't enable Mouser without a `key` (kicost.distributors - log.py:107)
WARNING:(WC017) Can't enable Elemen14 without a `key` (kicost.distributors - log.py:107)
WARNING:(WC017) Can't enable TME without a `token` and an `app_secret` (kicost.distributors - log.py:107)
WARNING:(WC017) Can't enable Nexar without a `client_id`, `client_secret` and `cache_path` (kicost.distributors - log.py:107)
Distributor list:

C:\Users\naman\Documents\KiCad\7.0>`

set-soft, 2023-06-22:
> i added the config file and the api key for mouser in the config.yaml file. however, it still gives WARNING:(WC017) Can't enable Mouser without a `key` (kicost.distributors - log.py:107)
> 

> `C:\Users\naman\Documents\KiCad\7.0>kicost -c "C:\Program Files\KiCad\7.0\.config\kicost\config.yaml" 

Here you must specify the input file.
Also note that you are storing your config in a place that KiCost won't find it unless you explicitly inform it. The file should be placed in `~/.config/kicost/` dir. Where `~` is the name path to the user home. Windows also have user home dirs.
Using `--debug 10` KiCost will say the name of the config file that is trying to load.

> WARNING:(WC017) Can't enable Digi-Key without a `client_id`, `client_secret`and`cache_path`(kicost.distributors - log.py:107) 

Ok, you just added Mouser.

> WARNING:(WC017) Can't enable Elemen14 without a`key`(kicost.distributors - log.py:107) 

Same

> WARNING:(WC017) Can't enable TME without a`token`and an`app_secret`(kicost.distributors - log.py:107) 

Same

> WARNING:(WC017) Can't enable Nexar without a`client_id`, `client_secret`and`cache_path` (kicost.distributors - log.py:107) 

Same, you are not getting an error about Mouser

> Traceback (most recent call last): File "C:\Program Files\KiCad\7.0\bin\Lib\runpy.py", line 197, in _run_module_as_main return _run_code(code, main_globals, None, File "C:\Program Files\KiCad\7.0\bin\Lib\runpy.py", line 87, in _run_code exec(code, run_globals) File "C:\Users\naman\Documents\KiCad\7.0\3rdparty\Python39\Scripts\kicost.exe__main__.py", line 7, in File "C:\Users\naman\Documents\KiCad\7.0\3rdparty\Python39\site-packages\kicost__main__.py", line 479, in main main_real() File "C:\Users\naman\Documents\KiCad\7.0\3rdparty\Python39\site-packages\kicost__main__.py", line 410, in main_real kicost_gui(args.force_en_us) # Use the user gui if no input is given. TypeError: kicost_gui() missing 4 required positional arguments: 'files', 'configure_kicost_apis', 'command_line_api_options', and 'args'

This is because you forgot to add an input file. This is fixed in git repo. I'll see if I can create a new release, but people should confirm the fix is working.

> C:\Users\naman\Documents\KiCad\7.0>kicost --show_dist_list 

Here you are not providing a config file, so you should get a warning for Mouser.

> WARNING:(WC017) Can't enable Digi-Key without a `client_id`, `client_secret` and `cache_path` (kicost.distributors - log.py:107) 
> WARNING:(WC017) Can't enable Mouser without a `key` (kicost.distributors - log.py:107) 

Which is logic because this time you didn't specify a config file.
The config file command line option must be specified every time you invoke KiCost, unless you move the file to the correct place.

set-soft, 2023-06-22:
KiCost 1.1.18 should fix the no arguments issue.
Open a different issue if you find another problem.

closed 2023-06-22 by set-soft