hildogjr/KiCost #547: pip and 23.04 can't install

mxa, 2023-05-29:
ok, so 23.04 seems to have done something to pip.

```
pip install kicost
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```

ok, so on stackexchange i find a recommendation to use pipx.

```
pipx install kicost
Fatal error from pip prevented installation. Full pip output in file:
    /home/mxa/.local/pipx/logs/cmd_2023-05-29_11.48.20_pip_errors.log

pip seemed to fail to build package:
    wxPython>=3.0.2

Some possibly relevant errors from pip install:
    error: subprocess-exited-with-error
    ModuleNotFoundError: No module named 'attrdict'
    error: metadata-generation-failed

Error installing kicost.
```

okay, in that logfile i read:

error: subprocess-exited-with-error
  
```
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [8 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-install-usoqlna8/wxpython_df15967db8654cdf864616f7ec488673/setup.py", line 27, in <module>
          from buildtools.config import Config, msg, opj, runcmd, canGetSOName, getSOName
        File "/tmp/pip-install-usoqlna8/wxpython_df15967db8654cdf864616f7ec488673/buildtools/config.py", line 30, in <module>
          from attrdict import AttrDict
      ModuleNotFoundError: No module named 'attrdict'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

```

set-soft, 2023-05-29:
> ok, so 23.04 seems to have done something to pip.

This is Python policy, not Debian policy (PEP 668). You can try doing what the explanation says, this is use `--break-system-packages`. But I'll suggest other solution: install it from the KiBot Debian repo: https://set-soft.github.io/debian/

After adding the repo and updating the packages list you'll be able to install the `kicost` Debian package.

IMHO pip (and all the Python module and package system) is poorly designed and strongly biased towards inefficient Windows usage. But this seems to be the *fashion of the moment*. Things are even worst in the Ruby and JavaScript front.

mxa, 2023-05-29:
nice, a debian package! however my config seems gone:

```
kicost
WARNING:(WC018) Unknown API `Octopart` (kicost - config.py:112)
WARNING:(WC017) Can't enable Digi-Key without a `client_id`, `client_secret` and `cache_path` (kicost.distributors - api_digikey.py:113)
WARNING:(WC017) Can't enable Elemen14 without a `key` (kicost.distributors - api_element14.py:253)
WARNING:(WC017) Can't enable TME without a `token` and an `app_secret` (kicost.distributors - api_tme.py:296)
WARNING:(WC017) Can't enable Nexar without a `client_id`, `client_secret` and `cache_path` (kicost.distributors - api_nexar.py:209)
Traceback (most recent call last):
  File "/home/mxa/.local/bin/kicost", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/lib/python3/dist-packages/kicost/__main__.py", line 479, in main
    main_real()
  File "/usr/lib/python3/dist-packages/kicost/__main__.py", line 410, in main_real
    kicost_gui(args.force_en_us)  # Use the user gui if no input is given.
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: kicost_gui() missing 4 required positional arguments: 'files', 'configure_kicost_apis', 'command_line_api_options', and 'args'

```

set-soft, 2023-05-29:
> nice, a debian package! however my config seems gone:

Where did you save it? Is it in *~/.config/kicost/*?

> File "/home/mxa/.local/bin/kicost", line 8, in <module>

Uninstall this stuff from pip, you are running an old version script that is trying to use the new kicost.
The simplest thing to try is to remove */home/mxa/.local/bin/kicost*, so you get the tool from */usr/bin/*.

mxa, 2023-06-06:
>  Where did you save it? Is it in ~/.config/kicost/?

Yes, there is the `config.yaml`

> Uninstall this stuff from pip, you are running an old version script that is trying to use the new kicost.
> The simplest thing to try is to remove /home/mxa/.local/bin/kicost, so you get the tool from /usr/bin/.

I've removed kicost there with `~/.local/bin$ rm kicost` and verified it's gone.
However, still getting the same 

```
WARNING:(WC018) Unknown API `Octopart` (kicost - config.py:112)
WARNING:(WC017) Can't enable Digi-Key without a `client_id`, `client_secret` and `cache_path` (kicost.distributors - api_digikey.py:113)
WARNING:(WC017) Can't enable Elemen14 without a `key` (kicost.distributors - api_element14.py:253)
WARNING:(WC017) Can't enable TME without a `token` and an `app_secret` (kicost.distributors - api_tme.py:296)
WARNING:(WC017) Can't enable Nexar without a `client_id`, `client_secret` and `cache_path` (kicost.distributors - api_nexar.py:209)
Traceback (most recent call last):
  File "/usr/local/bin/kicost", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/lib/python3/dist-packages/kicost/__main__.py", line 479, in main
    main_real()
  File "/usr/lib/python3/dist-packages/kicost/__main__.py", line 410, in main_real
    kicost_gui(args.force_en_us)  # Use the user gui if no input is given.
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: kicost_gui() missing 4 required positional arguments: 'files', 'configure_kicost_apis', 'command_line_api_options', and 'args'
```

set-soft, 2023-06-07:
> "/usr/local/bin/kicost"

You still have copies of KiCost at local places. You must ensure only one is installed.

mxa, 2023-06-07:
after removing that one too, I get: 
```
kicost
WARNING:(WC018) Unknown API `Octopart` (kicost - config.py:112)
WARNING:(WC017) Can't enable Digi-Key without a `client_id`, `client_secret` and `cache_path` (kicost.distributors - api_digikey.py:113)
WARNING:(WC017) Can't enable Elemen14 without a `key` (kicost.distributors - api_element14.py:253)
WARNING:(WC017) Can't enable TME without a `token` and an `app_secret` (kicost.distributors - api_tme.py:296)
WARNING:(WC017) Can't enable Nexar without a `client_id`, `client_secret` and `cache_path` (kicost.distributors - api_nexar.py:209)
Traceback (most recent call last):
  File "/usr/bin/kicost", line 33, in <module>
    sys.exit(load_entry_point('kicost==1.1.17', 'console_scripts', 'kicost')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kicost/__main__.py", line 479, in main
    main_real()
  File "/usr/lib/python3/dist-packages/kicost/__main__.py", line 410, in main_real
    kicost_gui(args.force_en_us)  # Use the user gui if no input is given.
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: kicost_gui() missing 4 required positional arguments: 'files', 'configure_kicost_apis', 'command_line_api_options', and 'args'
```

set-soft, 2023-06-08:
Ok, I see. Try calling KiCost using: **kicost --gui NAME_OF_THE_XML_NETLIST**

set-soft, 2023-06-08:
About the missing config options: run KiCost with *--debug 10* option to look from where is the config loaded.

ns2gupta, 2023-06-16:
> About the missing config options: run KiCost with _--debug 10_ option to look from where is the config loaded.

Hi, i am having the same issue with the config, I recently installed the KiCost using the documentation. 

after running kicost --debug i got this - 

C:\Users\naman\Documents\KiCad\7.0>kicost --debug
WARNING:(WC017) Can't enable Digi-Key without a `client_id`, `client_secret` and `cache_path` (kicost.distributors - log.py:107)
WARNING:(WC017) Can't enable Mouser without a `key` (kicost.distributors - log.py:107)
WARNING:(WC017) Can't enable Elemen14 without a `key` (kicost.distributors - log.py:107)
WARNING:(WC017) Can't enable TME without a `token` and an `app_secret` (kicost.distributors - log.py:107)
WARNING:(WC017) Can't enable Nexar without a `client_id`, `client_secret` and `cache_path` (kicost.distributors - log.py:107)
Traceback (most recent call last):
  File "C:\Program Files\KiCad\7.0\bin\Lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\KiCad\7.0\bin\Lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\naman\AppData\Local\Programs\Python\Python311\Scripts\kicost.exe\__main__.py", line 7, in <module>
  File "C:\Program Files\KiCad\7.0\bin\Lib\site-packages\kicost\__main__.py", line 479, in main
    main_real()
  File "C:\Program Files\KiCad\7.0\bin\Lib\site-packages\kicost\__main__.py", line 410, in main_real
    kicost_gui(args.force_en_us)  # Use the user gui if no input is given.
TypeError: kicost_gui() missing 4 required positional arguments: 'files', 'configure_kicost_apis', 'command_line_api_options', and 'args'

set-soft, 2023-06-22:
KiCost 1.1.18 should fix the no arguments issue

closed 2023-06-22 by set-soft

a-tawil, 2023-09-28:
I just installed 1.1.18 and still get the same warning of not finding distributors
`PS C:\Windows\system32> kicost -i "FILE.xml" --include mouser --overwrite --debug 10
DEBUG:No config file found (C:\Users\Ahmed Al-Tawil\.config\kicost\config.yaml) (kicost - log.py:115)
DEBUG:Final API options {'Local': {}, 'Digi-Key': {}, 'Mouser': {}, 'Element14': {}, 'TME': {}, 'Nexar': {}, 'KitSpace': {}} (kicost - log.py:115)
DEBUG:API options with defaults (kicost - log.py:115)
DEBUG:- Local: (kicost - log.py:115)
DEBUG:  - cache_ttl: 7 (kicost - log.py:115)
DEBUG:  - cache_path: C:\Users\Ahmed Al-Tawil/.cache/kicost\Local (kicost - log.py:115)
DEBUG:- Digi-Key: (kicost - log.py:115)
DEBUG:  - cache_ttl: 7 (kicost - log.py:115)
DEBUG:  - cache_path: C:\Users\Ahmed Al-Tawil/.cache/kicost\Digi-Key (kicost - log.py:115)
DEBUG:- Mouser: (kicost - log.py:115)
DEBUG:  - cache_ttl: 7 (kicost - log.py:115)
DEBUG:  - cache_path: C:\Users\Ahmed Al-Tawil/.cache/kicost\Mouser (kicost - log.py:115)
DEBUG:- Element14: (kicost - log.py:115)
DEBUG:  - cache_ttl: 7 (kicost - log.py:115)
DEBUG:  - cache_path: C:\Users\Ahmed Al-Tawil/.cache/kicost\Element14 (kicost - log.py:115)
DEBUG:- TME: (kicost - log.py:115)
DEBUG:  - cache_ttl: 7 (kicost - log.py:115)
DEBUG:  - cache_path: C:\Users\Ahmed Al-Tawil/.cache/kicost\TME (kicost - log.py:115)
DEBUG:- Nexar: (kicost - log.py:115)
DEBUG:  - cache_ttl: 7 (kicost - log.py:115)
DEBUG:  - cache_path: C:\Users\Ahmed Al-Tawil/.cache/kicost\Nexar (kicost - log.py:115)
DEBUG:- KitSpace: (kicost - log.py:115)
DEBUG:  - cache_ttl: 7 (kicost - log.py:115)
DEBUG:  - cache_path: C:\Users\Ahmed Al-Tawil/.cache/kicost\KitSpace (kicost - log.py:115)
DEBUG:Local API configured to enabled True (kicost.distributors - log.py:115)
WARNING:(WC017) Can't enable Digi-Key without a `client_id`, `client_secret` and `cache_path` (kicost.distributors - log.py:107)
DEBUG:Digi-Key API configured to enabled False id None secret None path C:\Users\Ahmed Al-Tawil/.cache/kicost\Digi-Key (kicost.distributors - log.py:115)
WARNING:(WC017) Can't enable Mouser without a `key` (kicost.distributors - log.py:107)
DEBUG:Mouser API configured to enabled False key None path C:\Users\Ahmed Al-Tawil/.cache/kicost\Mouser (kicost.distributors - log.py:115)
WARNING:(WC017) Can't enable Elemen14 without a `key` (kicost.distributors - log.py:107)
DEBUG:Element14 API configured to enabled False key None path C:\Users\Ahmed Al-Tawil/.cache/kicost\Element14 (kicost.distributors - log.py:115)
WARNING:(WC017) Can't enable TME without a `token` and an `app_secret` (kicost.distributors - log.py:107)
DEBUG:TME API configured to enabled False token None app_secret None path C:\Users\Ahmed Al-Tawil/.cache/kicost\TME (kicost.distributors - log.py:115)
WARNING:(WC017) Can't enable Nexar without a `client_id`, `client_secret` and `cache_path` (kicost.distributors - log.py:107)
DEBUG:Nexar API configured to enabled False id None secret None path C:\Users\Ahmed Al-Tawil/.cache/kicost\Nexar (kicost.distributors - log.py:115)
DEBUG:KitSpace API configured to enabled False (kicost.distributors - log.py:115)
ERROR:Unknown distributor requested: `mouser` [Known: []] (kicost - __init__.py:1475)`

a-tawil, 2023-09-28:
I just installed 1.1.18 and still get the same warning of not finding distributors
`PS C:\Windows\system32> kicost -i "FILE.xml" --include mouser --overwrite --debug 10 DEBUG:No config file found (C:\Users\Ahmed Al-Tawil\.config\kicost\config.yaml) (kicost - log.py:115) DEBUG:Final API options {'Local': {}, 'Digi-Key': {}, 'Mouser': {}, 'Element14': {}, 'TME': {}, 'Nexar': {}, 'KitSpace': {}} (kicost - log.py:115) DEBUG:API options with defaults (kicost - log.py:115) DEBUG:- Local: (kicost - log.py:115) DEBUG:  - cache_ttl: 7 (kicost - log.py:115) DEBUG:  - cache_path: C:\Users\Ahmed Al-Tawil/.cache/kicost\Local (kicost - log.py:115) DEBUG:- Digi-Key: (kicost - log.py:115) DEBUG:  - cache_ttl: 7 (kicost - log.py:115) DEBUG:  - cache_path: C:\Users\Ahmed Al-Tawil/.cache/kicost\Digi-Key (kicost - log.py:115) DEBUG:- Mouser: (kicost - log.py:115) DEBUG:  - cache_ttl: 7 (kicost - log.py:115) DEBUG:  - cache_path: C:\Users\Ahmed Al-Tawil/.cache/kicost\Mouser (kicost - log.py:115) DEBUG:- Element14: (kicost - log.py:115) DEBUG:  - cache_ttl: 7 (kicost - log.py:115) DEBUG:  - cache_path: C:\Users\Ahmed Al-Tawil/.cache/kicost\Element14 (kicost - log.py:115) DEBUG:- TME: (kicost - log.py:115) DEBUG:  - cache_ttl: 7 (kicost - log.py:115) DEBUG:  - cache_path: C:\Users\Ahmed Al-Tawil/.cache/kicost\TME (kicost - log.py:115) DEBUG:- Nexar: (kicost - log.py:115) DEBUG:  - cache_ttl: 7 (kicost - log.py:115) DEBUG:  - cache_path: C:\Users\Ahmed Al-Tawil/.cache/kicost\Nexar (kicost - log.py:115) DEBUG:- KitSpace: (kicost - log.py:115) DEBUG:  - cache_ttl: 7 (kicost - log.py:115) DEBUG:  - cache_path: C:\Users\Ahmed Al-Tawil/.cache/kicost\KitSpace (kicost - log.py:115) DEBUG:Local API configured to enabled True (kicost.distributors - log.py:115) WARNING:(WC017) Can't enable Digi-Key without a client_id, client_secretandcache_path(kicost.distributors - log.py:107) DEBUG:Digi-Key API configured to enabled False id None secret None path C:\Users\Ahmed Al-Tawil/.cache/kicost\Digi-Key (kicost.distributors - log.py:115) WARNING:(WC017) Can't enable Mouser without akey(kicost.distributors - log.py:107) DEBUG:Mouser API configured to enabled False key None path C:\Users\Ahmed Al-Tawil/.cache/kicost\Mouser (kicost.distributors - log.py:115) WARNING:(WC017) Can't enable Elemen14 without akey(kicost.distributors - log.py:107) DEBUG:Element14 API configured to enabled False key None path C:\Users\Ahmed Al-Tawil/.cache/kicost\Element14 (kicost.distributors - log.py:115) WARNING:(WC017) Can't enable TME without atokenand anapp_secret(kicost.distributors - log.py:107) DEBUG:TME API configured to enabled False token None app_secret None path C:\Users\Ahmed Al-Tawil/.cache/kicost\TME (kicost.distributors - log.py:115) WARNING:(WC017) Can't enable Nexar without aclient_id, client_secretandcache_path(kicost.distributors - log.py:107) DEBUG:Nexar API configured to enabled False id None secret None path C:\Users\Ahmed Al-Tawil/.cache/kicost\Nexar (kicost.distributors - log.py:115) DEBUG:KitSpace API configured to enabled False (kicost.distributors - log.py:115) ERROR:Unknown distributor requested:mouser [Known: []] (kicost - __init__.py:1475)`

set-soft, 2023-09-28:
> I just installed 1.1.18 and still get the same warning of not finding distributors

1.1.18 fixes the *missing 4 required positional arguments* issue, but isn't magic.

> DEBUG:Loading config from: C:\Users\Ahmed Al-Tawil\.config\kicost\config.yaml (kicost - log.py:115)

Here is the file where you MUST add *your* tokens, KiCost can't provide tokens for you. Not at least for free.