hildogjr/KiCost #551: module 'ssl' has no attribute 'wrap_socket' on python 3.12

mundodisco8, 2023-10-24:
#### Issue / Problem report
1. Installed the latest KiCost version (v1.1.18) in a clean virtual environment created with Python 3.12.
2. Ran `kicost -i hardware\MyProject.xml -c kicost.yaml --overwrite`
    1. My config only has API keys for digikey
       ```
       kicost:
        version: 1
        # Cache Time To Live in days, -1 is forever
        # Default is 7
        cache_ttl: 30
        # Base directory for the APIs caches
        cache_path: ~/.cache/kicost
      
      
        APIs:
          Digi-Key:
            # Digi-Key Client ID for a registered APP
            client_id: myID
            # Digi-Key Client Secret for a registered APP
            client_secret: mySecret
            # Use the sandbox server, doesn't count the usage, but returns old data
            sandbox: false
            # Only enabled if the client_id and client_secret are defined
            enable: true
            # Directory for the APIs caches
            cache_path: ~/.cache/kicost/Digi-Key
            # Exclude products offered by 3rd party associates (marketplace)
            exclude_market_place_products: true
       ```
    2. My Digikey app has this `https://localhost:8139/digikey_callback` as its OAUTH callback. I have no idea what this is or means, or if it's the best way to do it, but I had no idea about how to configure it, and followed [this guide](https://pypi.org/project/kicost-digikey-api-v3/).
4. KiCost shows the login page on the browser
5. KiCost shows the following error and exists.

```
- Digi-Key [api] (https://developer.digikey.com/)
WARNING:Oauth2 token storage does not exist or malformed, creating new. (kicost.distributors - log.py:107)
Progress:   0%|                                                                                                                                                                        | 0/13 [00:00<?, ?part/s]Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\joel.santos\OneDrive - Brompton Bicycle Ltd\Desktop\Gruffalo_Dimmer\venv\Scripts\kicost.exe\__main__.py", line 7, in <module>
  File "C:\Users\joel.santos\OneDrive - Brompton Bicycle Ltd\Desktop\Gruffalo_Dimmer\venv\Lib\site-packages\kicost\__main__.py", line 479, in main
    main_real()
  File "C:\Users\joel.santos\OneDrive - Brompton Bicycle Ltd\Desktop\Gruffalo_Dimmer\venv\Lib\site-packages\kicost\__main__.py", line 469, in main_real
    kicost(in_file=args.input, eda_name=args.eda,
  File "C:\Users\joel.santos\OneDrive - Brompton Bicycle Ltd\Desktop\Gruffalo_Dimmer\venv\Lib\site-packages\kicost\kicost.py", line 249, in kicost
    query_part_info(parts, dist_list, currency)
  File "C:\Users\joel.santos\OneDrive - Brompton Bicycle Ltd\Desktop\Gruffalo_Dimmer\venv\Lib\site-packages\kicost\kicost.py", line 76, in query_part_info
    get_dist_parts_info(parts, dist_list, currency)
  File "C:\Users\joel.santos\OneDrive - Brompton Bicycle Ltd\Desktop\Gruffalo_Dimmer\venv\Lib\site-packages\kicost\distributors\__init__.py", line 50, in get_dist_parts_info
    distributor_class.get_dist_parts_info(parts, dist_list, currency)
  File "C:\Users\joel.santos\OneDrive - Brompton Bicycle Ltd\Desktop\Gruffalo_Dimmer\venv\Lib\site-packages\kicost\distributors\distributor.py", line 166, in get_dist_parts_info
    solved = api.query_part_info(parts, list(remaining), currency)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\joel.santos\OneDrive - Brompton Bicycle Ltd\Desktop\Gruffalo_Dimmer\venv\Lib\site-packages\kicost\distributors\api_digikey.py", line 208, in query_part_info
    api_digikey._query_part_info(parts, distributors, currency)
  File "C:\Users\joel.santos\OneDrive - Brompton Bicycle Ltd\Desktop\Gruffalo_Dimmer\venv\Lib\site-packages\kicost\distributors\api_digikey.py", line 160, in _query_part_info
    data = o.search()
           ^^^^^^^^^^
  File "C:\Users\joel.santos\OneDrive - Brompton Bicycle Ltd\Desktop\Gruffalo_Dimmer\venv\Lib\site-packages\kicost_digikey_api_v3\utils.py", line 112, in search
    results = kicost_digikey_api_v3.manufacturer_product_details(body=search_request, api_limits=self.api_limit, **DK_API.extra_ops)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\joel.santos\OneDrive - Brompton Bicycle Ltd\Desktop\Gruffalo_Dimmer\venv\Lib\site-packages\kicost_digikey_api_v3\v3\api.py", line 131, in manufacturer_product_details
    client = DigikeyApiWrapper('manufacturer_product_details_with_http_info', kicost_digikey_api_v3.v3.productinformation)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\joel.santos\OneDrive - Brompton Bicycle Ltd\Desktop\Gruffalo_Dimmer\venv\Lib\site-packages\kicost_digikey_api_v3\v3\api.py", line 50, in __init__
    self._digikeyApiToken = kicost_digikey_api_v3.oauth.oauth2.TokenHandler(version=3, sandbox=self.sandbox).get_access_token()
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\joel.santos\OneDrive - Brompton Bicycle Ltd\Desktop\Gruffalo_Dimmer\venv\Lib\site-packages\kicost_digikey_api_v3\oauth\oauth2.py", line 276, in get_access_token
    httpd.socket = ssl.wrap_socket(httpd.socket, certfile=filename, server_side=True)
                   ^^^^^^^^^^^^^^^
[... 212 lines cut ...]
    data = o.search()
           ^^^^^^^^^^
  File "C:\Users\joel.santos\OneDrive - Brompton Bicycle Ltd\Desktop\Gruffalo_Dimmer\venv12\Lib\site-packages\kicost_digikey_api_v3\utils.py", line 112, in search
    results = kicost_digikey_api_v3.manufacturer_product_details(body=search_request, api_limits=self.api_limit, **DK_API.extra_ops)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\joel.santos\OneDrive - Brompton Bicycle Ltd\Desktop\Gruffalo_Dimmer\venv12\Lib\site-packages\kicost_digikey_api_v3\v3\api.py", line 131, in manufacturer_product_details
    client = DigikeyApiWrapper('manufacturer_product_details_with_http_info', kicost_digikey_api_v3.v3.productinformation)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\joel.santos\OneDrive - Brompton Bicycle Ltd\Desktop\Gruffalo_Dimmer\venv12\Lib\site-packages\kicost_digikey_api_v3\v3\api.py", line 50, in __init__
    self._digikeyApiToken = kicost_digikey_api_v3.oauth.oauth2.TokenHandler(version=3, sandbox=self.sandbox).get_access_token()
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\joel.santos\OneDrive - Brompton Bicycle Ltd\Desktop\Gruffalo_Dimmer\venv12\Lib\site-packages\kicost_digikey_api_v3\oauth\oauth2.py", line 276, in get_access_token
    httpd.socket = ssl.wrap_socket(httpd.socket, certfile=filename, server_side=True)
                   ^^^^^^^^^^^^^^^
AttributeError: module 'ssl' has no attribute 'wrap_socket'
Progress:   0%|
```

6. Write the command used to call KiCost (or the graphical interface configuration);

`kicost -i hardware\MyProject.xml -c kicost.yaml --overwrite`

7. One BoM to reproduce the error (with the EDA version).

[Example.zip](https://github.com/hildogjr/KiCost/files/13120384/Example.zip) from Kicad 7.0.7

mundodisco8, 2023-10-26:
I took the liberty of messing with the kicost-digikey-api-v3 code and managed to get it working, but I'm not very good at this so more work might be needed.

https://github.com/set-soft/kicost-digikey-api-v3/pull/1

Feel free to reject if it's not good enough.

set-soft, 2023-11-08:
Hi @mundodisco8 !
The patch is now merged and included in v0.1.3 release.

closed 2023-11-08 by set-soft