hildogjr/KiCost #552: Update kicost v1.1.6

AlanRavelo, 2023-11-16:
Hi. I have installed version kicost v.1.1.6 on Windows 10 for a long time and it works well performing the BOM. For a month now I have not been able to perform the BOM. I think maybe I need to update.
In the image I get this error: 

Starting cost processing ...
ERROR:Internal error: HTTPSConnectionPool(host='dev-partinfo.kitspace.org', port=443): Max retries exceeded with url: /graphql (Caused by SSLError(SSLError(1, u'[SSL: REUSE_CERT_TYPE_NOT_ZERO] unsolicited extension (_ssl.c:727)'),)) (kicost - kicost_gui.py:151)

[image]

set-soft, 2023-11-16:
KitSpace is down. It finished working months ago. You need to use a new version.

Do you have Digi-Key and Mouser credentials to use them? If not you'll need them

closed 2023-11-16 by set-soft

AlanRavelo, 2023-11-16:
hi, you can perform the update but you need to insert the distributor's
api.
I got my api key for the mouser distributor,
where do I have to insert it to start kicost? thank you

Alan Ravelo

AlanRavelo, 2023-11-17:
Hi, I already made the update.
I got my API key for Mouser distrbuitor,
I have tried to insert it but without success. Where do I have to insert it to start kicost? thanks for your answer.

set-soft, 2023-11-17:
Please consult the [docs](https://hildogjr.github.io/KiCost/docs/_build/singlehtml/index.html#document-configuration)

AlanRavelo, 2023-11-21:
Hi, I appreciate your help.
  I installed version 1.1.18 from the "power shell" console but that "config.yaml" file does not appear. Do I have to create that file with the text? .
Manually download the version 1.1.18 file from gihub and the text appears in the file: KiCost-1.1.18\tests\configs\mouser. Do I have to copy it to the folder where it was installed using the "power shell" console?

I don't understand the "Configuration file" stage very well (https://hildogjr.github.io/KiCost/docs/_build/singlehtml/index.html#document-configuration)

set-soft, 2023-11-21:
> Hi, I appreciate your help. I installed version 1.1.18 from the "power shell" console but that "config.yaml" file does not appear. Do I have to create that file with the text? . 

Yes, you must create it.

> Manually download the version 1.1.18 file from gihub and the text appears in the file: KiCost-1.1.18\tests\configs\mouser. Do I have to copy it to the folder where it was installed using the "power shell" console?

You can just use the template found in the docs, just modify the *Mouser* section to include your *key*

> I don't understand the "Configuration file" stage very well (https://hildogjr.github.io/KiCost/docs/_build/singlehtml/index.html#document-configuration)

Is just a text file. The format is YAML, this is almost plain text. The *#* is used for comments. You have to remove the `# ` before *key* in the *Mouser* section and replace *XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX* using the key you got from Mouser.

If you don't know where to copy the file just run `kicost --debug 10`, the first debug message will say where KiCost is trying to read the configuration file.

AlanRavelo, 2023-11-22:
Hi, thanks for your help!!, I managed to configure the api keys for kicost.
I get the file with the prices of the products.

Only when enabling the "digi-key" dealer option do I get an error as shown in the image. Insert the two keys ("client_id" and "client_secret") but I am hesitant to add "OAuth Callback". When I run, a webb window opens. Will you know why it is?
Only when enabling the "digi-key":
[image]

a webb window opens:
[image]

set-soft, 2023-11-22:
The browser is opened to get the OAuth token from Digi-Key.

pvolkert, 2023-12-04:
> Hi, thanks for your help!!, I managed to configure the api keys for kicost. I get the file with the prices of the products.
> 
> Only when enabling the "digi-key" dealer option do I get an error as shown in the image. Insert the two keys ("client_id" and "client_secret") but I am hesitant to add "OAuth Callback". When I run, a webb window opens. Will you know why it is? Only when enabling the "digi-key": <img alt="prueba_03" width="881" src="https://private-user-images.githubusercontent.com/88397949/284752405-58b534f4-6123-4a06-9c30-bdff27717774.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE2OTc5OTEsIm5iZiI6MTcwMTY5NzY5MSwicGF0aCI6Ii84ODM5Nzk0OS8yODQ3NTI0MDUtNThiNTM0ZjQtNjEyMy00YTA2LTljMzAtYmRmZjI3NzE3Nzc0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjA0VDEzNDgxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk1NDNiYzlhNjQwMDNkMjY5MDAwN2Q4ZjgyYzc0MDRhNDVlZDA5NTk3NzNkZjYxOTAzYzY0ZTFjNjZiNWM1NjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.g0elslher3jmNEfiBPokhQPN1qJeojuA9XNKnVjKWGg">
> 
> a webb window opens: <img alt="prueba_04" width="755" src="https://private-user-images.githubusercontent.com/88397949/284752412-d0f811d0-98b1-404b-b457-cc68b7681e51.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE2OTc5OTEsIm5iZiI6MTcwMTY5NzY5MSwicGF0aCI6Ii84ODM5Nzk0OS8yODQ3NTI0MTItZDBmODExZDAtOThiMS00MDRiLWI0NTctY2M2OGI3NjgxZTUxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjA0VDEzNDgxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE0NmZlZTgwNGVhMzUxNjQ1YTM5OTE5NmZmY2M0YjY0N2ZjMTlmMTFiMGNkZDk5NWM2YWQ0MmZmMGUzNDFiYWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.wt10PTgUDkPnH6vZ489cQswHPI0Jmu0Oz404nGMmgd0">

Same problem here with Digikey, have you been able solve it? I tried also activating the different APIs in my digikey account, but I always get the same error as shown by you.

pvolkert, 2023-12-04:
Thanks for the information! I tried adding the callback as mentioned and I see this also in the URL:

> https://sandbox-api.digikey.com/v1/oauth2/authorize?client_id=THISISMYID&response_type=code&redirect_uri=https%3A%2F%2Flocalhost%3A8139%2Fdigikey_callback

so this seems to be fine for my feeling. Unfortunately the problem is still the same, identical error message as before.

Also I tried using the "kicost-digikey-api-v3" by installing and running the test-script as provided in your repo. This fails due to a missing "configure" function:

>Traceback (most recent call last):
>  File "C:\Users\xxx\temp\kicost-digikey-api-v3\test_production.py", line 1, in <module>  from kicost_digikey_api_v3 import by_digikey_pn, by_keyword, by_manf_pn, configure
> ImportError: cannot import name 'configure' from 'kicost_digikey_api_v3' (C:\Users\xxx\temp\kicost-digikey-api-v3\kicost_digikey_api_v3\__init__.py)

I wasn't able to find the function in the repo, seems not to be the one from DK_API. Any idea what I am doing wrong?

pvolkert, 2023-12-04:
I tried a little further and I see the following problem or is it a misconfiguration?:

The error-page is always shown by the sandbox-api, no matter if the "sandbox" key in the yaml is set to true or false. Is there a way to force the program to use the production API?

I now created a sandbox api client, with this it takes data from the server (also with the sandbox key set to true or false). The data itself are not really useful, just seem to be some random price data, often the same for totally different parts.

set-soft, 2023-12-05:
> I wasn't able to find the function in the repo, seems not to be the one from DK_API. Any idea what I am doing wrong?

The example was outdated, thanks for reporting the problem.
I pushed a new version with the missing code (no longer part of the API).

> I tried a little further and I see the following problem or is it a misconfiguration?:
> 
> The error-page is always shown by the sandbox-api, no matter if the "sandbox" key in the yaml is set to true or false. Is there a way to force the program to use the production API?

The sandbox-api is used only when you specify its use. Make sure the environment variable *DIGIKEY_CLIENT_SANDBOX* isn't defined as *True*. And yes, I got errors trying a very old sandbox key I used in the past. But the regular API worked perfectly.

> I now created a sandbox api client, with this it takes data from the server (also with the sandbox key set to true or false). The data itself are not really useful, just seem to be some random price data, often the same for totally different parts.

If the sandbox-api server is used even when KiCost YAML says `sandbox: false` then you have the environment variable defined.

pvolkert, 2023-12-05:
> > I wasn't able to find the function in the repo, seems not to be the one from DK_API. Any idea what I am doing wrong?
> 
> The example was outdated, thanks for reporting the problem. I pushed a new version with the missing code (no longer part of the API).

Thanks, this works now fine, also with the production API.

But the problem with KiCost remains for me.
I checked your debug messages and found this what looks for me a little bit confusing:

First at startup, sandbox = false is recognized correctly from the yaml:

> DEBUG:Final API options {'Digi-Key': {'client_id': 'Exxxx', 'client_secret': 'jxxxx', 'sandbox': False, 'exclude_market_place_products': False}, 'Mouser': {'key': 'kkk'}, 'Local': {}, 'Element14': {}, 'TME': {}, 'Nexar': {}, 'KitSpace': {}} (kicost - log.py:115)

This is also shown in the GUI log window:
> DEBUG:- Digi-Key: (kicost - log.py:115)
> DEBUG:  - client_id: EXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXt (kicost - log.py:115)
> DEBUG:  - client_secret: jXXXXXXXXXXXXXXU (kicost - log.py:115)
> DEBUG:  - sandbox: False (kicost - log.py:115)
> DEBUG:  - exclude_market_place_products: False (kicost - log.py:115)

When starting the BOM creation, in the textbox the following is written:
>DEBUG:Overwriting Digi-Key.cache_path = C:\Users\xxx\.cache\kicost\Digi-Key (kicost.distributors - log.py:115)
>DEBUG:**Overwriting Digi-Key.sandbox** = True (kicost.distributors - log.py:115)
>DEBUG:Overwriting Digi-Key.client_id = EXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXt (kicost.distributors - log.py:115)
> DEBUG:Final API options {'Digi-Key': {'client_id': 'Exxxx', 'client_secret': 'jxxx', **'sandbox': True**, 'exclude_market_place_products': False, 'cache_path': 'C:\\Users\\xxx\\.cache\\kicost\\Digi-Key'}, 'Mouser': {'key': 'dxxx'}, 'Local': {'cache_ttl': 1, 'cache_path': 'C:\\Users\\xxx\\.cache\\kicost\\Local'}, ....

KiCost accordingly tries to open the sandbox API, what for sure fails as the credentials are for the production environment.

I also deleted everything in the cache. Did not change anything.

Any idea why the key from my config file is overwritten?

set-soft, 2023-12-05:
> Any idea why the key from my config file is overwritten?

An environment variable is the source for this message. You have *DIGIKEY_CLIENT_SANDBOX* defined to *True*

pvolkert, 2023-12-06:
I'm sorry if i'm too stupid but I can't solve the issure... 

I can't see any environment variable with this name, neither using `set` or using the windows (11) gui.

I now even created an environment variable (global, see screenshot below) `DIGIKEY_CLIENT_SANDBOX=false` (also gave it a try with 0), the problem still is the same.

[image]

I also removed and reinstalled the digikey-api-v3 package, problem remains the same :(

set-soft, 2023-12-06:
> I'm sorry if i'm too stupid but I can't solve the issue...
> 
> I can't see any environment variable with this name, neither using `set` or using the windows (11) gui.

Double check the debug output, which configuration file is loaded? Take a look at the file, are you sure it doesn't ask for sandbox?

Try specifying the configuration file from the command line (*--config*). This will disable the environment overwrite.

The code is using *os.getenv*, so you have it defined somewhere.