hildogjr/KiCost #553: ERROR:Internal error: 'rohsStatusCode' (kicost - kicost_gui.py:150)

mxa, 2023-11-19:
#### Issue / Problem report
KiCost v1.1.18

I am getting this output:
```

Traceback (most recent call last):
  File "/usr/bin/kicost", line 33, in <module>
    sys.exit(load_entry_point('kicost==1.1.18', 'console_scripts', 'kicost')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kicost/__main__.py", line 479, in main
    main_real()
  File "/usr/lib/python3/dist-packages/kicost/__main__.py", line 469, in main_real
    kicost(in_file=args.input, eda_name=args.eda,
  File "/usr/lib/python3/dist-packages/kicost/kicost.py", line 249, in kicost
    query_part_info(parts, dist_list, currency)
  File "/usr/lib/python3/dist-packages/kicost/kicost.py", line 76, in query_part_info
    get_dist_parts_info(parts, dist_list, currency)
  File "/usr/lib/python3/dist-packages/kicost/distributors/__init__.py", line 50, in get_dist_parts_info
    distributor_class.get_dist_parts_info(parts, dist_list, currency)
  File "/usr/lib/python3/dist-packages/kicost/distributors/distributor.py", line 166, in get_dist_parts_info
    solved = api.query_part_info(parts, list(remaining), currency)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kicost/distributors/api_element14.py", line 369, in query_part_info
    api_element14._query_part_info(dist, country, parts, distributors, currency)
  File "/usr/lib/python3/dist-packages/kicost/distributors/api_element14.py", line 308, in _query_part_info
    specs = {'rohs': ('RoHS', data['rohsStatusCode'])}
                              ~~~~^^^^^^^^^^^^^^^^^^
KeyError: 'rohsStatusCode'
```

set-soft, 2023-11-21:
Hi @mxa !
Thanks for reporting, can you check that the above patch solves your problem?

closed 2023-11-21 by set-soft

mxa, 2023-11-22:
Yes, that fixed it. Thanks!