hildogjr/KiCost #557: KiCad v8 BOM generation changed (no XML file), how to export to KiCost?

craigarno, 2024-02-07:
- New KiCad v8 BOM Tool
- No more XML file for input.
[image]

[image]

#### Issue / Problem report
1. P:\Project\555\KiCAD8>kicost -v
KiCost v1.1.18

2. Add a terminal error report / txt, use ``--debug`` option on KiCost call;
    kicost -w -i %YOUR_FILE% --debug 8
3. I usually use:
kicost -w -i "555.xml" --eda kicad --fields Description

Now there is only the ability to produce CSV's, no XML is produced.

What is needed to easily import KiCad v8 BOM CSV's? Here is an example file:
[555.csv](https://github.com/hildogjr/KiCost/files/14191546/555.csv)

4. When I try to import, I get this output:
```
P:\Project\555\KiCAD8>kicost -w -i "555.csv" --eda kicad --fields Description
WARNING:(WC018) Unknown API `Octopart` (kicost - log.py:107)
WARNING:(WC001) "description" field is a reserved field and can not be used as user field. Try to remove it from internal dictionary using `--translate_fields description ~` (kicost - log.py:107)
- Digi-Key [api] (https://developer.digikey.com/)
Progress:   0%|                                                                                | 0/9 [00:00<?, ?part/s]Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Program Files\KiCad\8.0\bin\Scripts\kicost.exe\__main__.py", line 7, in <module>
  File "C:\Program Files\KiCad\8.0\bin\Lib\site-packages\kicost\__main__.py", line 479, in main
    main_real()
  File "C:\Program Files\KiCad\8.0\bin\Lib\site-packages\kicost\__main__.py", line 469, in main_real
    kicost(in_file=args.input, eda_name=args.eda,
  File "C:\Program Files\KiCad\8.0\bin\Lib\site-packages\kicost\kicost.py", line 249, in kicost
    query_part_info(parts, dist_list, currency)
  File "C:\Program Files\KiCad\8.0\bin\Lib\site-packages\kicost\kicost.py", line 76, in query_part_info
    get_dist_parts_info(parts, dist_list, currency)
  File "C:\Program Files\KiCad\8.0\bin\Lib\site-packages\kicost\distributors\__init__.py", line 50, in get_dist_parts_info
    distributor_class.get_dist_parts_info(parts, dist_list, currency)
  File "C:\Program Files\KiCad\8.0\bin\Lib\site-packages\kicost\distributors\distributor.py", line 166, in get_dist_parts_info
    solved = api.query_part_info(parts, list(remaining), currency)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\KiCad\8.0\bin\Lib\site-packages\kicost\distributors\api_digikey.py", line 208, in query_part_info
    api_digikey._query_part_info(parts, distributors, currency)
  File "C:\Program Files\KiCad\8.0\bin\Lib\site-packages\kicost\distributors\api_digikey.py", line 160, in _query_part_info
    data = o.search()
           ^^^^^^^^^^
  File "C:\Program Files\KiCad\8.0\bin\Lib\site-packages\kicost_digikey_api_v3\utils.py", line 114, in search
    DK_API.cache.save_results('mpn', self.manf_pn, results)
  File "C:\Program Files\KiCad\8.0\bin\Lib\site-packages\kicost\distributors\distributor.py", line 86, in save_results
    with open(self.get_name(prefix, name), "wb") as fh:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 22] Invalid argument: 'C:\\Users\\craig\\.cache\\kicost\\Digi-Key\\mpn_Reference\tValue\tDatasheet\tFootprint\tQty\tDNP_US_en_USD_US.dat'
Progress:   0%|
```

#### Feature / Enhancement request
- Soon v8 will become official and be out.  Many other's will run into this issue.

set-soft, 2024-02-07:
Hi @craigarno !
My advice: wait some months until KiCad 8 becomes stable, just take a look at the (huge) amount of bugs in 6.0.0 and 7.0.0.

Right now I can't even find a Debian/Ubuntu package of 8.0.0rc2, may be I can use a docker image, but again be patient. KiCad is developed in a way that plug-in break, compatibility is not in the KiCad team agenda. I'm sure you can generate the XML in some way.

craigarno, 2024-02-07:
I was able to half answer my own question:
[image]
"Legacy" still generates XML for KiCost.  I can't live without KiCost, so this is an upgrade dealbreaker to me if KiCost needs an XML and KiCad can't produce an XML (thankfully, it still does, from the menu).

Anyway, yes, it took me ½ year before I could accept v7 into my work flow.

There are two features which really stand out to me as nice in v8 which make me want to adopt...

- New Schematic PDF generation with added component information for support people (think purchasing, maybe manufacturing) who don't have and don't want KiCad.

[image]

And new SPICE enhancements which make it easier to work with modeling (trivial example)
[image]

Anyway, v8 is coming, but now I can breathe again.

Hey, thank you for all your support @set-soft !

set-soft, 2024-02-07:
> * New Schematic PDF generation with added component information for support people (think purchasing, maybe manufacturing) who don't have and don't want KiCad.

Note that for people wanting to see KiCad projects, without KiCad, the KiCanvas project is a promising thing.

You can see how KiBot can generate it [here](https://inti-cmnb.github.io/kibot_variants_arduprog_site/Browse/t1-kicanvas.html), you can also see other KiBot stuff [here](https://inti-cmnb.github.io/kibot_variants_arduprog_site/Browse/t1-navigate.html)

set-soft, 2024-03-22:
Currently I have KiCad 8.0.1 docker images working, but they use unstable Linux. KiCad people didn't create packages for an Ubuntu compatible with the current Debian stable, don't know why because the nightly builds included "Lunar", which is compatible.

Anyways, I think this issue is solved, so we can close it.

closed 2024-03-22 by set-soft