hildogjr/KiCost #560: Digikey api error

BrianVod, 2024-03-04:
I am using Kicad 7, and trying to use the kicost for Digikey costing as well as other distributers. All the distributers I am using seem to work fine and print everything to the Excel spread flawlessly. However, the Digikey run, errors out with the error:
"ERROR:Internal error 'NoneType' object has no attribute 'products' (kicost-_int_.py:1568)" I am not sure how to correct this.  This is trying to access a production api on digikey. I also tried the sandbox and got the same error. The api is : api_digikey.py.

So I am looking for some adivice on how to find this error.
Thanks in Advance.

set-soft, 2024-03-05:
> "ERROR:Internal error 'NoneType' object has no attribute 'products' (kicost-_int_.py:1568)"

I'm not sure which source code is mentioned in the above text. Can you provide a screen capture? (What is _int_?)

BrianVod, 2024-03-05:
I am not sure what "int" I am confused by that.

Here is the entire file: 

# -*- coding: utf-8 -*-
# MIT license
#
# Copyright (C) 2021 by Salvador E. Tropea / Instituto Nacional de Tecnologia Industrial
#
# Permission is hereby granted, free of charge, to any person obtaining a copy
# of this software and associated documentation files (the "Software"), to deal
# in the Software without restriction, including without limitation the rights
# to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
# copies of the Software, and to permit persons to whom the Software is
# furnished to do so, subject to the following conditions:
#
# The above copyright notice and this permission notice shall be included in
# all copies or substantial portions of the Software.
#
# THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
# IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
# FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
# AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
# LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
# OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
# THE SOFTWARE.

# Author information.
__author__ = 'Salvador Eduardo Tropea'
__webpage__ = 'https://github.com/set-soft'
__company__ = 'Instituto Nacional de Tecnologia Industrial - Argentina'

# Libraries.
import pprint

# KiCost definitions.
from .. import DistData, KiCostError, W_NOINFO, ERR_SCRAPE, W_APIFAIL
# Distributors definitions.
from .distributor import distributor_class, QueryCache, hide_secrets
from .log__ import debug_detailed, debug_overview, debug_obsessive, warning

available = True
try:
    from kicost_digikey_api_v3 import by_digikey_pn, by_manf_pn, by_keyword, DigikeyError, DK_API
except ImportError:
    available = False

    class DK_API(object):
        api_ops = None

DIST_NAME = 'digikey'
# Specs known by KiCost
SPEC_NAMES = {'tolerance': 'tolerance',
              'power (watts)': 'power',
              'voltage - rated': 'voltage',
              'manufacturer': 'manf',
              'size / dimension': 'size',
              'temperature coefficient': 'temp_coeff',
              'frequency': 'frequency',
              'package / case': 'footprint'}
__all__ = ['api_digikey']

class api_digikey(distributor_class):
    name = 'Digi-Key'
    type = 'api'
    # Currently enabled only by request
    enabled = available
    url = 'https://developer.digikey.com/'  # Web site API information.
    api_distributors = [DIST_NAME]
    # Options supported by this API
[... 123 lines cut ...]
                if value:
                    dd.extra_info['value'] = value
                for spec, name in SPEC_NAMES.items():
                    val = specs.get(spec, None)
                    if val:
                        dd.extra_info[name] = val[1]
                part.dd[DIST_NAME] = dd
                debug_obsessive('* Part info after adding data:')
                debug_obsessive(pprint.pformat(part.__dict__))
                debug_obsessive(pprint.pformat(dd.__dict__))
            progress.update(1)
        progress.close()

    @staticmethod
    def query_part_info(parts, distributors, currency):
        msg = None
        try:
            api_digikey._query_part_info(parts, distributors, currency)
        except DigikeyError as e:
            msg = e.args[0]
        if msg is not None:
            raise KiCostError(msg, ERR_SCRAPE)
        return set([DIST_NAME])

distributor_class.register(api_digikey, 100)

BrianVod, 2024-03-05:
Here is the full error:

Starting cost processing ...

- Digi-Key [api] (https://developer.digikey.com/)

ERROR:Internal error: 'NoneType' object has no attribute 'products' (kicost - __init__.py:1568)

 

Let me know if you need anything else.

Thanks

--Brian

set-soft, 2024-03-06:
I fail to find any  `__init__.py` file with line 1568. Try running KiCost with the `--debug 10` option.

BrianVod, 2024-03-06:
Ok Ill give that a try. Thanks for looking into this.

BrianVod, 2024-03-07:
Ok, I finally got to run the debug and here is what it says:

 

ERROR:In configuration file: Error loading YAML while parsing a block mapping

  in "C:\Users\brian\.config\kicost\config.yaml", line 12, column 3

expected <block end>, but found '<block mapping start>'

  in "C:\Users\brian\.config\kicost\config.yaml", line 16, column 6 (kicost - __init__.py:1568)

 

So it looks to be something in the config file. I used the sample on the site as a guide but I must have something incorrect.

 

Here is my config file text:

 

# KiCost configuration file

kicost:

  version: 1

  # Cache Time To Live in days, -1 is forever

  # Default is 7

  cache_ttl: -1

  # Base directory for the APIs caches

  # cache_path: ~/.cache/kicost

 

 

APIs:

  Digi-Key:

   #Digi-Key Client ID for a registered APP

   client_id:'id'

    #Digi-Key Client Secret for a registered APP

     client_secret: 'secret'

    # Use the sandbox server, doesn't count the usage, but returns old data

     sandbox: false

    # Only enabled if the client_id and client_secret are defined

     enable: true

    # Directory for the APIs caches

     cache_path: ~/.cache/kicost/Digi-Key

    # Exclude products offered by 3rd party associates (marketplace)

     exclude_market_place_products: false

  Element14:

    # Element14 includes: Farnell, Newark and CPC

    # Element14 Product Search API key

     key: 'key'

    # Only enabled if the key is defined

     enable: True

    # Country used for Farnell queries.

    # Supported countries: BG,CZ,DK,AT,CH,DE,IE,IL,UK,ES,EE,FI,FR,HU,IT,LT,

    # LV,BE,NL,NO,PL,PT,RO,RU,SK,SI,SE,TR,CN,AU,NZ,HK,SG,MY,PH,TH,IN,KR,VN

    # farnell_country: UK

    # Country used for Newark queries.

     Supported countries: US,CA,MX

     newark_country: US

    # Country used for CPC queries.

    # Supported countries: UK,IE

    # cpc_country: UK

    # Directory for the APIs caches

     cache_path: ~/.cache/kicost/Element14

  Mouser:

    # Mouser Part API key

     key: a1c0ce7f-546f-473a-9b21-f815a4bd26f6

    # Only enabled if the key is defined

     enable: True

    # Directory for the APIs caches

     cache_path: ~/.cache/kicost/Mouser

  Nexar:

    # Nexar client ID

     client_id: 36cb4a25-ffc4-4613-b869-cbd2c9fefa87

    # Nexar client secret

     client_secret: 8JycdKSpgvvuy39gmHs4CzDrTbHhJps8TaAG

    # Only enabled if the client_id and client_secret are defined

     enable: True

    # Country where we are buying

     country: US

    # Directory for the APIs caches

     cache_path: ~/.cache/kicost/Nexar

  TME:

    # TME token (anonymous or private)

    # token: XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

    # TME application secret

    # app_secret: XXXXXXXXXXXXXXXXXXXX

    # Only enabled if the token and app_secret are defined

    # enable: false

    # Country where we are buying

    # country: US

    # Language for the texts

    # language: EN

    # Directory for the APIs caches

    # cache_path: ~/.cache/kicost/TME

BrianVod, 2024-03-07:
A little more information for you:

If I do not use the config file then Kicost runs and Farnell, Mouser and Newark run fine and populate the CBOM as it should.

But when running with Digikey in the mix then it fails, also when I use NEXAR it fails with  :

- Nexar [api] (https://nexar.com/api)

ERROR:Internal error: 'data' (kicost - __init__.py:1568)

 

So to sum up whats happening;

With the config file:

1.	Kicost does not start with the config file and debug shows the error I sent earlier

 

Without the config file:

1.	Kicost Runs 
2.	Digikey Fails 
3.	NEXAR Fails : ERROR:Internal error: 'data' (kicost - __init__.py:1568)
4.	Newark Runs
5.	Farnell Runs
6.	Mouser Runs

 

It seemed at first NEXAR worked, then it stopped. Is there a way to uninstall Kicost and start over as if only Kicad exits? I tried “pip uninstall Kicost, it said it removed it but everything was still there. I tried reinstalling over the top but that didn’t change anything.

 

If I could uninstall it I would try a fresh install and see of that helps.

 

Thanks,

--Brian

set-soft, 2024-03-07:
> Ok, I finally got to run the debug and here is what it says: ERROR:In configuration file: Error loading YAML while parsing a block mapping

Indentation of YAML files is important. Your configuration file is broken, please read about YAML format.

Note that in order to get some cost you need a valid configuration file.

BrianVod, 2024-03-07:
The config file seems to be working now but I still get an error when I enable NEXAR. I copied the error from Kicost. If I don’t enable the nexar api then I can get costs from Newark, Mouser, Farnell, which tells me element14 api is working.. Digikey still not working but I think that’s on their end. 

 

“Starting cost processing ...

- Nexar [api] (https://nexar.com/api)

ERROR:Internal error: 'data' (kicost - __init__.py:1568)”

 

I don’t where or what this could be coming from.

Let me know if need something from me to review and I will get it to you.

Thanks,

--Brian

BrianVod, 2024-03-07:
one other note, what I also getting is this error in the terminal:
WARNING:(WC018) Unknown option `Element14 includes` for API `Element14` (kicost - log.py:107)
WARNING:(WC018) Unknown option `Supported countries` for API `Element14` (kicost - log.py:107)

closed 2024-03-18 by BrianVod