hildogjr/KiCost #562: Digikey API: pyopenssl 24.1.0 breaking a

Andrew-Collins, 2024-03-25:
- Use a simple and clear title.
- Do not mark as "ISSUE", "FEATURE" on title, this will be tagged by the development group.

#### Issue / Problem report
1. Add the KiCost version. Type ``kicost --version`` at terminal.
2. Add a terminal error report / txt, use ``--debug`` option on KiCost call;
    kicost -w -i %YOUR_FILE% --debug 8
3. Write the command used to call KiCost (or the graphical interface configuration);
4. One BoM to reproduce the error (with the EDA version).

#### Feature / Enhancement request
- Explain how it will help the other users.

The team thanks your contribution.

Andrew-Collins, 2024-03-25:
Thanks again for your work on these tools, if you would like I should have some time on Friday have crack at fixing this.

set-soft, 2024-03-26:
Hi @Andrew-Collins !
The newest I have in my docker images is 24.0.0.
Note that the traceback says the problem is just doing: `from OpenSSL import crypto`, so it might be just some library version mismatch.

Andrew-Collins, 2024-03-27:
Hi @set-soft,

This error will only occur on 'fresh' installs as the `kicost-digikey-api-v3` package does not restrict the version of `pyopenssl` it requires; so if `pyopenssl` is already installed then the build will use it as is, instead of pulling the latest from `pypi`.

If you build in a fresh venv, uninstall `pyopenssl`, or possibly rebuild your docker image it should start using the lastest `pyopenssl` version.

Andrew-Collins, 2024-03-27:
This brings up the point that if we elect to make changes, that the `kicost-digikey-api-v3` package will need to specify `pyopenssl >= 24.1.0`...

set-soft, 2024-03-27:
Hi @Andrew-Collins !

I run my "experimental" docker images (used for KiCad 8), and upgraded the package to 24.1.0, I then reinstalled the digikey API package. It installs and works.

So again: take a look at the traceback, the crash is inside the *pyopenssl* code and triggered by just requesting the crypto functionality (`from OpenSSL import crypto`). IMHO this is a package error. The libcrypto is most probably missing the *X509_V_FLAG_NOTIFY_POLICY* symbol. I don't know if your package is using a system level libcrypto, or one shipped with the python code (something regrettable, but very Pythonic), in any case I bet the version of libcrypto isn't the correct one.

If you find a problem in the code please open a new issue. Here I don't see anything we can do from the digikey API side.

closed 2024-03-27 by set-soft

Andrew-Collins, 2024-03-27:
Hi @set-soft,

I'm very sorry, I misread and thought you were talking about the `pyopenssl` version mismatch :facepalm:

I was originally having this problem while building in a fresh venv, but I just tried again and was unable to replicate this error. Unsure what changed :upside_down_face:  

And for anyone else who may find this, my system level install was failing due to old versions of `pyopenssl` and `cryptography` that were installed at the system level.