hildogjr/KiCost #564: KiCost "bulk orders" not working in LibreOffice Calc, does work in MS Excel with Digikey and Mouser

craigarno, 2024-04-09:
```
Edition	Windows 11 Pro
Version	23H2
Installed on	‎10/‎11/‎2022
OS build	22631.3296
Serial number	03073...
Experience	Windows Feature Experience Pack 1000.22687.1000.0

```
kicad>kicost --version
**KiCost v1.1.18**

I have a KiCost 1.1.18 Spreadsheet, and I am trying to order parts in bulk after filling in the "Purch" field.

Copy/Paste to DigiKey and Mouser works in MS-Excel but doesn't work in LibreOffice Calc.

When I copy the contents of the Bulk parts field, I get something different in **Excel** than I get from **Calc:**

**Excel**
"10,BAT-HLD-001-ND,B1
20,399-T521D685M050ATE070CT-ND,C1;C2
200,399-C1210C476M4PAC7800CT-ND,C3;C15;C16;C24-C40
80,455-BM04B-SRSS-TBCT-ND,J27-J34
30,RMCF1206JG1K50CT-ND,R21-R23
10,732-13663-1-ND,SW1
30,36-5002-ND,TP1;TP5;TP6
10,175-MAXM15067AMB+TCT-ND,U1
20,742-SIC967ED-T1-GE3CT-ND,U2;U3
"

**Calc**
10,BAT-HLD-001-ND,B1 20,399-T521D685M050ATE070CT-ND,C1;C2 200,399-C1210C476M4PAC7800CT-ND,C3;C15;C16;C24-C40 80,455-BM04B-SRSS-TBCT-ND,J27-J34 30,RMCF1206JG1K50CT-ND,R21-R23 10,732-13663-1-ND,SW1 30,36-5002-ND,TP1;TP5;TP6 10,175-MAXM15067AMB+TCT-ND,U1 20,742-SIC967ED-T1-GE3CT-ND,U2;U3 

The Bulk Order Entry works in Excel only.

LibreOffice Calc looks like the issue might be Linux line endings [lf] endings vs Windows [cr][lf].  When I correct line endings by hand, then bulk entry works again.  In a parts list with 65 different parts, this hand editing of line endings becomes tedious.

frankalicious, 2025-04-18:
I have the same issue in Linux.  
As a workaround I replaced in the formula of the bulk parts field
```
=CONCATENATE(
```
with
```
=TEXTJOIN(CHAR(13);1;
```

---
Copied output with CONCATENATE
```
603-RC0603FR-0749K9L|10|R301 603-RC0603FR-1320K5L|10|R302 
```

Copied output with TEXTJOIN
```
603-RC0603FR-0749K9L|10|R301 
603-RC0603FR-1320K5L|10|R302 
```

craigarno, 2025-04-22:
First, thanks @frankalicious for offering a solution!

And as always, big thanks to @set-soft for maintaining this product... I use it on all my projects.

I tried the 1.20 mainline install, and it still has CONCATENATE which

I then tried @frankalicious solution with "=TEXTJOIN(CHAR(13);1;" and it produces the desirable individual lines under Windows and in LibreOffice Calc, as suggested.

Then I decided it would be good to get all the latest changes with the latest commit and received this error:

```
C:\Users\craig\Desktop\Catamaran\tiller-board-main\kicad>pip install git+https://github.com/xesscorp/KiCost.git
Collecting git+https://github.com/xesscorp/KiCost.git
  Cloning https://github.com/xesscorp/KiCost.git to c:\users\craig\appdata\local\temp\pip-req-build-gopwcr80
  Running command git clone --filter=blob:none --quiet https://github.com/xesscorp/KiCost.git 'C:\Users\craig\AppData\Local\Temp\pip-req-build-gopwcr80'
  Resolved https://github.com/xesscorp/KiCost.git to commit 0753c8d88332289caca75a9b25f78edd3adbfe1c
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [16 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "C:\Users\craig\AppData\Local\Temp\pip-req-build-gopwcr80\setup.py", line 7, in <module>
          import kicost.version
        File "C:\Users\craig\AppData\Local\Temp\pip-req-build-gopwcr80\kicost\__init__.py", line 17, in <module>
          from .kicost import query_part_info, solve_parts_qtys  # noqa F401
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\craig\AppData\Local\Temp\pip-req-build-gopwcr80\kicost\kicost.py", line 61, in <module>
          from .edas.tools import field_name_translations, subpartqty_split, group_parts, PRJ_STR_DECLARE, PRJPART_SPRTR, partgroup_qty
        File "C:\Users\craig\AppData\Local\Temp\pip-req-build-gopwcr80\kicost\edas\__init__.py", line 27, in <module>
          from .eda_kicad import eda_kicad  # noqa: F401
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\craig\AppData\Local\Temp\pip-req-build-gopwcr80\kicost\edas\eda_kicad.py", line 29, in <module>
          from bs4 import BeautifulSoup
      ModuleNotFoundError: No module named 'bs4'
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```
@set-soft something still to resolve in the latest commit before I can use it.

frankalicious, 2025-04-23:
> I tried the 1.20 mainline install, and it still has CONCATENATE which

Yes, you need to use the git version.  

Have you tried to remove kicost and then use the git version together with the upgrade option?  
```
pip uninstall kicost
pip install --upgrade git+https://github.com/hildogjr/KiCost.git
```

set-soft, 2025-04-23:
Hi @craigarno !
The fixed code uses `CONCATENATE` it just changes the EOL (End Of Line) character (from Line Feed to Carriage Return), which seems to be handled by the three major spreadsheets.

> BTW, I'll be at KiCon 2025 in San Diego if either of you are going.

Good luck! Is too expensive for me ...

closed 2025-04-23 by set-soft