hildogjr/KiCost #567: Starting cost processing ... ERROR:Internal error: invalid literal for int() with base 10: '' (kicost - kicost_gui.py:150)

mxa, 2024-06-20:
#### Issue / Problem report
KiCost v1.1.19 on Kubuntu 24.04 I'm getting this error:

```
DEBUG:Getting parts... (kicost.edas - generic_csv.py:197)
Traceback (most recent call last):
  File "/usr/bin/kicost", line 33, in <module>
    sys.exit(load_entry_point('kicost==1.1.19', 'console_scripts', 'kicost')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kicost/__main__.py", line 479, in main
    main_real()
  File "/usr/lib/python3/dist-packages/kicost/__main__.py", line 469, in main_real
    kicost(in_file=args.input, eda_name=args.eda,
  File "/usr/lib/python3/dist-packages/kicost/kicost.py", line 172, in kicost
    p, info = get_part_groups(eda_name[i_prj], in_file[i_prj], ignore_fields, variant[i_prj], dist_list)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kicost/edas/__init__.py", line 49, in get_part_groups
    return eda_class.get_part_groups(eda, in_file, ignore_fields, variant, distributors)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kicost/edas/eda.py", line 119, in get_part_groups
    parts, prj_info = eda_class.registered[eda].get_part_groups(in_file, distributors)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kicost/edas/generic_csv.py", line 223, in get_part_groups
    return get_part_groups(in_file, distributors)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kicost/edas/generic_csv.py", line 204, in get_part_groups
    refs, fields, gen_cntr = extract_fields(row, header, header_file, dialect, gen_cntr)
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kicost/edas/generic_csv.py", line 93, in extract_fields
    refs = split_refs(ref_str)
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kicost/edas/tools.py", line 777, in split_refs
    split = list(range(int(split_nums[0]), int(split_nums[1])+1))
                       ^^^^^^^^^^^^^^^^^^
ValueError: invalid literal for int() with base 10: ''

```

set-soft, 2024-06-24:
You are most probably using an invalid reference for a component.
You must use "XXXNNN" references, i.e. "D1", not things like "I'm a testpoint".
You can consult IPC standards about what a circuit is supposed to use

closed 2024-06-24 by set-soft

mbuchwald-atg, 2024-07-12:
Had the same issue while testing manually exported data from Altium that I formatted according to the example BOM here: https://github.com/hildogjr/KiCost/files/1480151/generic_BOM_example.zip
It doesn't use XXXNNN  designators exclusively in the second column, so I just exported my component description into the second column, enclosed in quotes (same behaviour without quotes most of the time, I'd say), and expected it to work. After all, the example uses fields like "U2 (alt., low offset)".

Works fine until the script trips over an alphanumeric block followed by a minus sign, e.g.

```
Mfr. No;Customer No;Order Qty.
"1";"yarp-75";13´<-- works
"2";"yarp-75-1";13 <-- works
"3";"yarp-TN";13 <-- works
"4";"yarp-A75-1";13 <-- works
"5";"yarp-75A-1";13 <-- works
"6";"yarp-75TN48C";13 <-- works
"7";"75yarp-75";13 <-- works ERROR:Internal error: list index out of range (kicost - kicost_gui.py:150) 
"8";"foo75yarp-75";13 <--ERROR:Internal error: invalid literal for int() with base 10: '' (kicost - kicost_gui.py:150)
"9";"yarp75-TN";13 <-- ERROR:Internal error: invalid literal for int() with base 10: '' (kicost - kicost_gui.py:150)
"10";"yarp75N-TN";13 <-- ERROR:Internal error: invalid literal for int() with base 10: '' (kicost - kicost_gui.py:150)
"11";"yarp75N-123";13 <-- ERROR:Internal error: invalid literal for int() with base 10: '' (kicost - kicost_gui.py:150)
```

Sure, if that isn't what the script expects in this column, it's on us - I'd still prefer a better error handling, as e.g. an additional empty line at the very end of the CSV is mentioned, but this totally breaks data processing. These two errors are 100% non-obvious to someone just trying to get off the ground with a small sample dataset after successfully using the example for distributor setup testing.
If a strict XXXNNN naming scheme is required - why not regex over the data and throw an error message right away?

set-soft, 2024-07-16:
Hi @mbuchwald-atg !

I didn't write the CSV import code, it looks like the code tries to be smart about component ranges, but I don't have a complete list of cases to be covered, so I can't fully fix it.

The 244617d58aeeec94e89ff2f4e9b94edde1151668 patch adds a couple of sanity checks, it can load the test cases you mentioned. Please let me know if it works for you.