hildogjr/KiCost #569: Throttle Farnell queries to avoid Server Error Forbidden 403 (403 Developer Over QPS (Proxy))

mbuchwald-atg, 2024-07-15:
#### Issue / Problem report
1. KiCost v1.1.19
2. see attached logfile
3. kicost --debug 8 -w -i allparts.csv
4. Any BOM will work, as long as there are more than two uncached part numbers in it that are fetched within one second. Example:

```
Mfr. No;Customer No;Order Qty.
MCMET 2.5A 250V;F26;1
MCMEF 5A 250V;F5;1
178.6165.000;F6;1
0154001.DRT;F8;1
015401.5DRT;F9;1
6214-0-15-15-21-27-10-0;J30;1
0461-3-15-15-21-27-04-0;J32;1
0552-2-57-15-21-27-10-0;J36;1
0918 520 6324;JP1;1
0918 516 7913;JP10;1
87427-0202;JP100;1
GMKDS 3/ 2-7,62;JP101;1
```

Base Farnell product search API access allows for 2 calls per second and 1000 calls per day. Currently, the script fetching data doesn't seem rate limited, so once it gets going and no cached entries slow it down, it can easily exceed 2 calls per second. The server then responds with a 403 which kills the script - fetching can be restarted immediately (manual action!), but worst case it will only download two more datasets before the next 403 hits.
Terrible user experience, total waste of processing power to get back to the point where the script just terminated, and also likely not ideal usage of limited API calls (unknown if a blocked one still counts as used).
The script already measures retrieved parts per second, so we should make this acceptable for Farnell before KiCost is widely adopted and Farnell decides to take specific countermeasures or slows down free API access even more.

Top 5 Error Response Codes from the API website:  (yeah, that's not five, innit?)
```
Date/Time,"403 Developer Over QPS (Proxy)"
2024-07-09T00:00:00.000
2024-07-10T00:00:00.000
2024-07-11T00:00:00.000
2024-07-12T00:00:00.310
2024-07-13T00:00:00.000
2024-07-14T00:00:00.200
2024-07-15T00:00:00.000
```

Top 5 Method Calls from the same website:
```
Date/Time,Successful,Blocked,Other
2024-07-09T00:00,0,0,0
2024-07-10T00:00,0,0,0
2024-07-11T00:00,0,0,0
2024-07-12T00:00,205,31,0
2024-07-13T00:00,0,0,0
2024-07-14T00:00,226,20,0
2024-07-15T00:00,0,0,0
```

[kicost.log](https://github.com/user-attachments/files/16231608/kicost.log)

set-soft, 2024-07-16:
Hi @mbuchwald-atg !
Thanks for reporting, I couldn't reproduce it because my queries are slow from Argentina.
I tested the code with a much slower speed limit (2 queries in 6 seconds) and it slowed the queries accordingly.
Please let me know if it works for you.

BTW: The "GMKDS 3/ 2-7,62" was split by KiCost, something I think you don't want

closed 2024-07-16 by set-soft

mbuchwald-atg, 2024-07-22:
Hi,
just came around testing the fix: Works really well 👍
Values smaller than the suggested 0.5 (seconds) in the `while (datetime.datetime.now() - self.last_query).total_seconds() < 0.5:` line still trigger the 403 forbidden error within a couple queries, so that seems a strictly enforced limit by Farnell. The debug output for throttling shows up constantly :)
Thank you very much for the quick fix!

Regarding unexpected part number things: I'm currently feeding our entire Altium database of~1.5k MPNs into KiCad (and therefore KiCost), and have been fixing, updating and replacing a couple items each that caused problems. MPNs with non-escaped commas are an issue, I just don't know yet where to fix them, as in "KiCad should do that", "KiCost should sanitize inputs" or "manufacturers shouldn't issue silly item codes". I'll probably open a couple more reports once everything is tidied up and I know more about the root causes of their respective issues. Regardless, #569 seems fixed and should be included in the next release. Thanks!

commit b1aaf71e (2024-07-16): [Element14][Fixed] Too fast queries

Throttle Farnell queries to avoid Server Error Forbidden 403
(403 Developer Over QPS (Proxy))

Fixes #569