hildogjr/KiCost #570: kicost 1.1.19 crashes on Macos Sonoma 14.6.1

pwittich, 2024-08-23:
`kicost` crashes after successfully running on Macos Sonoma 14.6.1, using an anaconda environment. I can reproduce the crash w/o even processing a file (see below). "Successfully running" means that the xlsx file is generated and is readable by Excel. So it is not an issue per se, though it is maybe concerning.

See below.

```
% kicost --version
KiCost v1.1.19
[1]    26140 segmentation fault  kicost --version
% python --version
Python 3.12.4
% kicost --version --debug 8          
KiCost v1.1.19
[1]    26295 segmentation fault  kicost --version --debug 8
```
Installed via pip in the virtual environment; 
```
% pip list
Package               Version
--------------------- -----------
beautifulsoup4        4.12.3
certifi               2024.7.4
cffi                  1.17.0
charset-normalizer    3.3.2
colorama              0.4.6
cryptography          43.0.0
filelock              3.15.4
idna                  3.7
inflection            0.5.1
kibom                 1.9.1
kicost                1.1.19
kicost-digikey-api-v3 0.1.3
lxml                  5.3.0
pillow                10.4.0
pip                   24.2
pycparser             2.22
pyOpenSSL             24.2.1
python-dateutil       2.9.0.post0
PyYAML                6.0.2
requests              2.32.3
requests-file         2.1.0
setuptools            72.1.0
six                   1.16.0
soupsieve             2.6
tldextract            5.1.2
tqdm                  4.66.5
urllib3               2.2.2
validators            0.33.0
wheel                 0.43.0
wxPython              4.2.1
XlsxWriter            3.2.0
```

set-soft, 2024-08-26:
Hi @pwittich !
Python code can't generate *segmentation fault* by itself. This is an issue with the Python interpreter and/or some lowl level library

pwittich, 2024-08-26:
ok, well, here is the output from `lldb`. It's in the cleanup of python itself.
```
(elec) [dev_board] which python                                                          7:17:51 
/Users/wittich/miniconda3/envs/elec/bin/python
(elec) [dev_board] lldb `which python`                                                   7:18:53 
(lldb) target create "/Users/wittich/miniconda3/envs/elec/bin/python"
Current executable set to '/Users/wittich/miniconda3/envs/elec/bin/python' (arm64).
(lldb) run /Users/wittich/miniconda3/envs/elec/bin/kicost --version
Process 37036 launched: '/Users/wittich/miniconda3/envs/elec/bin/python' (arm64)
KiCost v1.1.19
Process 37036 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x10)
    frame #0: 0x00000001000f1ce0 python`_PyObject_Free + 40
python`:
->  0x1000f1ce0 <+40>: ldr    x19, [x8, #0x10]
    0x1000f1ce4 <+44>: adrp   x8, 1092
    0x1000f1ce8 <+48>: ldr    x8, [x8, #0x7f8]
    0x1000f1cec <+52>: cmp    x8, x19
Target 0: (python) stopped.
(lldb) 
```
Anyhow, if you want to close this, go ahead.