hildogjr/KiCost #573: oauth2 error

mxa, 2024-10-01:
When running kicost v1.1.19 I get the error:
`Failed to use refresh token, starting new authorization flow. (kicost.distributors - oauth2.py:265) `
then the program stalls at 1%

What do I need to do?

set-soft, 2024-10-02:
Make sure you can start a web browser.
This is Digi-Key asking you to manually login to get a fresh token, is the only distributor using this annoying mechanism.
If you don't see a web browser opening then check what is wrong, i.e. if you run it on a docker image and you don't export the display and/or you don't have enough rights (needs to run `xhost +`)

mxa, 2024-10-03:
running `xdg-open index.html` will open the browser from the terminal. I don't know why kicost can't do it. I tried running it as sudo, but it doesn't change anything. It's not installed in a docker.

mxa, 2024-10-04:
Solved it. `/usr/bin/firefox` was pointing to the snap package which was not installed, so the python module `webbrowser` could not find it. I created an alias in /usr/bin/ to point to the binary, then it worked. Sorry for the noise.

closed 2024-10-04 by mxa