hildogjr/KiCost #578: Digikey not working anymore (error 401)

nguyen-v, 2025-03-19:
I've been using KiCost with KiBot, but it seems like it has stopped working recently (it was working before). I've setup the API key according to:
https://github.com/set-soft/kicost-digikey-api-v3

I think it might be an issue with the (deprecated?) API V3, as digikey now uses V4. I also found this issue from another project: https://github.com/peeter123/digikey-api/issues/53

```
 - Digi-Key [api] (https://developer.digikey.com/)
ERROR:Exception when calling manufacturer_product_details_with_http_info: (401)
Reason: Unauthorized
```

set-soft, 2025-03-20:
I tried right now, removing my token and forcing to get a new one, and it works.

I got errors from TME (API issue) and Element14 (unable to verify its SSL certificate).

Please check that this isn't a problem with your API key.

set-soft, 2025-03-20:
BTW: are you running it locally? If not: don't forget to refresh the token, it expires, digi-key isn't good for CI/CD workflows.

gjvanderheiden, 2025-03-21:
It really, really, really doesn't work your users say...

gjvanderheiden, 2025-03-21:
See opening post

gjvanderheiden, 2025-03-21:
Hi, there's been a PR for the digikey quite some time now. The digikey api doesn't get updates anymore.

I created a simple python script, which uses the digikey api project in a basic form, no banana. Doesn't work. Updated with the PR mentioned in the opening post, works. Tried a lot, deleting the cache dir etc. It is genuinely broken. [Ki-nTree](https://github.com/sparkmicro/Ki-nTree) also uses a branch, because of the maintenance issue. See their [requirements.txt](https://github.com/sparkmicro/Ki-nTree/blob/main/requirements.txt).

I haven't noticed your project before, came across it because of github notification of a mention of the issue in digikey-api. First I mistakenly thought it was a reaction on the digikey-api issue. Good to see projects like this.

set-soft, 2025-03-21:
> Hi, there's been a PR for the digikey quite some time now. The digikey api doesn't get updates anymore.
>
> I created a simple python script, which uses the digikey api project in a basic form, no [...]

But this is because you are using an API key with access to the V4 API, so you won't be able to use the V3 API.

This original issue was created because @nguyen-v said "Digikey not working anymore (error 401)", and this is not the case. API V3 is working, so @nguyen-v must check other sources of problems.

I entered to the API management interface and realized that Digikey is no longer giving V3 access. As I have keys with access to V3 API I can access it, but a new user will be forced to choose "Product Information V4" API, and here comes the problem. The code only supports V3.

The PR is interesting but must be reworked because it just replaces V3 by V4, which IMHO is a very bad idea. I checked and the status of my V3 keys is "active" and with no expiration day. So old users have V3 keys and no reason to create new ones. For this reason the version must be selectable.

> I haven't noticed your project before, came across it because of github notification of a mention of the issue in digikey-api. First I mistakenly thought it was a reaction on the digikey-api issue. Good to see projects like this.
> 
> Edit: Really good idea to use wx-python, already installed with KiCad, instead of yet another framework. I which Ki-nTree did that.

KiCad plug-ins should use wxPython, this helps to integrate them with KiCad, you get the same style of dialogs from a plug-in. It also helps, with installation, as you mention,  you know wxPython is there, KiCad includes it. I guess Ki-nTree author wanted to use something better than wxPython ... but I agree the cost isn't a good trade-off.

I'll take a look at the code from the PR to see if I can get V4 support added.

set-soft, 2025-03-27:
Ok, I released v1.1.20 with DigiKey API V4 support.
It needs a new Python package that implements it.
Also note that you MUST explicitly indicate ``version: 4` in the configuration file. People using v3 credentials doesn't need to change anything.
@gjvanderheiden : Can you try it?
@nguyen-v : Is this related to the original issue?

nguyen-v, 2025-03-27:
> Ok, I released v1.1.20 with DigiKey API V4 support.
> It needs a new Python package that implements it.
> Also note that you MUST explicitly indicate ``version: 4` in the configuration file. People [...]

Sorry I haven't had the time to try your suggestions yet, I will report back soon

gjvanderheiden, 2025-03-28:
@nguyen-v  Sorry, I wasn't totally clear, I don't use this software, I was just trying to help

nguyen-v, 2025-04-02:
> BTW: are you running it locally? If not: don't forget to refresh the token, it expires, digi-key isn't good for CI/CD workflows.

@set-soft How do I go about doing that? I've tried the following https://developer.digikey.com/tutorials-and-resources/oauth-20-3-legged-flow

I first did a code request with https://api.digikey.com/v1/oauth2/authorize?
response_type=code&client_id=123456789abcdefg
&redirect_uri=https%3A%2F%2Fclient-app-redirect-uri%2F using my client_id and redirect_uri, which got me the code.

I then used curl to send an authorization request, which got me the refresh_token, and then a refresh token request.

```
curl --request POST \
  --url https://api.digikey.com/v1/oauth2/token \
  --header 'Content-Type: application/x-www-form-urlencoded' \
  --data-urlencode 'client_id=YOUR_CLIENT_ID' \
  --data-urlencode 'client_secret=YOUR_CLIENT_SECRET' \
  --data-urlencode 'code=AUTHORIZATION_CODE' \
  --data-urlencode 'grant_type=authorization_code' \
  --data-urlencode 'redirect_uri=YOUR_REDIRECT_URI'
```

```
curl --request POST \
  --url https://api.digikey.com/v1/oauth2/token \
  --header 'Content-Type: application/x-www-form-urlencoded' \
  --data-urlencode 'client_id=YOUR_CLIENT_ID' \
  --data-urlencode 'client_secret=YOUR_CLIENT_SECRET' \
  --data-urlencode 'refresh_token=YOUR_REFRESH_TOKEN' \
  --data-urlencode 'grant_type=refresh_token'
```

I tried to run KiBot after this with KiCost, but it still returns the following:
```
ERROR:Exception when calling manufacturer_product_details_with_http_info: (401)
Reason: Unauthorized
```

```
HTTP response body: {"ErrorResponseVersion":"3.0.0.0","StatusCode": 401,"ErrorMessage":"Invalid Client-Id","ErrorDetails":"You are not subscribed to this API. Please subscribe and try again."
```

I've set up my production app as follows:
[image]

[image]

What am I doing wrong? Sorry I'm not very familiar with how this is supposed to be done. Should I delete and recreate a production app?

set-soft, 2025-04-03:
Hi @nguyen-v !
I always get the token manually and upload it to the project as a secret. If the token expires I use a new one.
Where are you storing the token? Are you creating `token_storage_v4.json`?

mxa, 2025-04-16:
Digikey works for me

Peugot206, 2026-02-02:
For some reason this works when I run it like `kicost -i 'file'`, it then succesfully opens a webpage to retreive the code. But when I try it through the GUI it doesn't work, as it can't seem to retreive the auth code. Maybe it's coincidence?

Well I'm happy there is way (for me) to get it working!