hildogjr/KiCost #579: TME: Bad Request

mxa, 2025-04-16:
I am getting this on TME:

```
- TME [api] (https://developers.tme.eu/en/)
ERROR:Server error `Bad Request` (400) (kicost - __main__.py:481)

```

set-soft, 2025-04-16:
Yes, TME is not working. I don't have much time to investigate it.

mxa, 2025-04-16:
It works!

_Dear User, Anonymous keys have been disabled. To fully utilize the functionality of the TME API, please use a private key._ 

One needs to go to  https://developers.tme.eu/dev
there add your TME customer account (may be different from the developer account for the API) and put the new key in the config.yaml file.

closed 2025-04-16 by mxa

set-soft, 2025-04-22:
Yes the problem was this.
Just to document the issue in case others face the same problem:

1. You must be sure you have a customer account at TME, this is in addition to your developer account
2. Login to the development account 
3. Locate the "Generate new private key" in your registered application
4. Fill the "Enter TME Login" with your TME **customer** user name (not necessarily the same as development account, and use the user, not e-mail)
5. Fill the "Enter TME password" with your TME **customer** password
6. Press the button
7. Use the Application secret and generated token