hildogjr/KiCost #582: [QUESTION] How does KiCost choose what part to return?

ymich9963, 2025-08-28:
I've been examining the source code and I just can't find anywhere how KiCost chooses to return a part. For example, for the part with MPN '1N4148' there are a lot of options on Mouser for that but it returns the one with the highest stock. Is there a way to control what gets returned? Is there a way of specifying how to prioritise what parts to return?

set-soft, 2025-09-01:
Nope, you can just select which one manually, by providing the distributor P/N.
There are too much details involved in selecting the best option, and the data from different distributors is different.
Also: when their belong to different manufacturers an automatic selection is dangerous, buying a cheap component from a poor manufacturer is not a good idea.

ymich9963, 2025-09-01:
Of course the last point goes without saying. I was just curious if there was a way of specifying through KiCost. Is there a way of specifying through their provided APIs?

set-soft, 2025-09-02:
> I see now that it is not possible through KiCost as it only takes the first part result,

This is not the same for each distributor. And also: some distributors sort the results using some criteria.

> Maybe a feature to add in the future?

Feel free to add a feature request. I don't have plans to implement it in the near future.

> I know this is for Mouser but I'm assuming this is the behaviour for the rest of the distributors for consistency.

As I mentioned above each API implementation is different.